elParaguayo/RPi-InfoScreen-Kivy #4: Squeezeplayer

elParaguayo, 2015-09-20:
Not started.

Basic: Allow control/visualisation of squeezeplayers.

elParaguayo, 2015-10-04:
Screenshots:
[image]
[image]

CraigHissett, 2015-10-04:
Great work!

elParaguayo, 2015-10-05:
This is a beast of  screen! Things to do before uploading:
- [ ] Update playlist if changed
- [ ] Select players
- [ ] Control volume

CraigHissett, 2015-10-05:
Keep going man - amazing work!!

elParaguayo, 2015-10-09:
I need a break from this for a while. Doing my head in. It's a bit rough around the edges but it works ok. Lots to improve on though.

elParaguayo, 2015-10-21:
Good enough for release!

closed 2015-10-21 by elParaguayo