elParaguayo/RPi-InfoScreen-Kivy #20: Images on weather screen flashing

elParaguayo, 2015-09-28:
Not sure how to fix this one... don't know if it's a bug in Kivy or an issue with the icons themselves.

Needs investigating.

elParaguayo, 2015-09-28:
Doing some digging. This is likely due to the fact that the icons are gifs. Setting the [anim_delay](http://kivy.org/docs/api-kivy.uix.image.html#kivy.uix.image.Image.anim_delay) property may be a simple solution for this.

elParaguayo, 2015-09-29:
Nope. That gave me lots of "critical" warnings!

elParaguayo, 2015-09-30:
Looks like it's a bug in Kivy itself: https://github.com/kivy/kivy/issues/2508

kivitalo, 2016-07-29:
If you install pillow and start the RPi-InfoScreen with following command:
`KIVY_IMAGE=pil python main.py` 
the flashing stops and icons are shown correctly.

elParaguayo, 2016-07-30:
Good to know, thank you.

elParaguayo, 2016-07-30:
Presumably we could add this to the main script:
```
import os
os.environ['KIVY_IMAGE'] = 'pil'
```

kivitalo, 2016-07-31:
I added that line to the main.py and it worked.

elParaguayo, 2016-08-01:
Ok. I'll include when I do next updates.

regit, 2016-08-03:
I confirm the fix.

elParaguayo, 2016-12-10:
This fix crashes the script on my laptop. Admittedly, that's not the target system but it's not ideal!