elParaguayo/RPi-InfoScreen-Kivy #22: Screen: Word Clock

elParaguayo, 2015-10-01:


elParaguayo, 2015-10-09:
Starting this one as I wanted a break from the SqueezePlayer one! This should be easier.

I know those words will come back to haunt me...

elParaguayo, 2015-10-09:
Screenshots:
[image]
[image]

elParaguayo, 2015-10-09:
Need to fix "teno'clock". Can do this by using the "ten" minutes at this point.

CraigHissett, 2015-10-10:
Love it!

elParaguayo, 2015-10-10:
Ok - this is done.

Last commit did the following:
 - Code tidy-up with comments
 - Added ability to add additional language layouts
 - Added fade transition between times

closed 2015-10-10 by elParaguayo