elParaguayo/RPi-InfoScreen-Kivy #41: Agenda app not working due to unicode error (plus solution)

LaurensBelgium, 2016-01-09:
Thank you for your amazing program which I am now using for quite some time.
I noticed that in the agenda functionality was having difficulties in handling non ASCI characters. (see error log)
This can be solved by adding support for string encoding in utf-8 by adding the following lines (see below). Afterwards the agenda functionality works normally.

%%%% add following code to screen.py file in the agenda folder %%%%%%

import sys
reload(sys)
sys.setdefaultencoding("utf-8")

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

errorlog
Traceback (most recent call last):
   File "./main.py", line 52, in <module>
     InfoScreenApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 828, in run
     runTouchApp()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 487, in runT                                                                     ouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.                                                                     py", line 89, in mainloop
     self._mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.                                                                     py", line 84, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 327, in idle
     Clock.tick()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 515, in tic                                                                     k
     self._process_events()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 647, in _pr                                                                     ocess_events
     event.tick(self._last_tick, remove)
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 406, in tic                                                                     k
     ret = callback(self._dt)
   File "/usr/local/lib/python2.7/dist-packages/kivy/animation.py", line 339, in                                                                      _update
     self.stop(widget)
   File "/usr/local/lib/python2.7/dist-packages/kivy/animation.py", line 209, in                                                                      stop
     self.dispatch('on_complete', widget)
   File "kivy/_event.pyx", line 714, in kivy._event.EventDispatcher.dispatch (ki                                                                     vy/_event.c:7654)
   File "kivy/_event.pyx", line 1224, in kivy._event.EventObservers.dispatch (ki                                                                     vy/_event.c:13497)
   File "kivy/_event.pyx", line 1148, in kivy._event.EventObservers._dispatch (k                                                                     ivy/_event.c:13023)
   File "/usr/local/lib/python2.7/dist-packages/kivy/uix/screenmanager.py", line                                                                      413, in _on_complete
     self.screen_in.dispatch('on_enter')
   File "kivy/_event.pyx", line 718, in kivy._event.EventDispatcher.dispatch (ki                                                                     vy/_event.c:7699)
   File "./screens/agenda/screen.py", line 112, in on_enter
     self.update()
   File "./screens/agenda/screen.py", line 136, in update
     self.drawCalendars()
   File "./screens/agenda/screen.py", line 309, in drawCalendars
     ev = CalendarItem(event=event)
   File "./screens/agenda/screen.py", line 52, in __init__
     self.formatEvent(kwargs["event"])
   File "./screens/agenda/screen.py", line 75, in formatEvent
     evtime = "{} - {}".format(evtime, event["location"])
 UnicodeEncodeError: 'ascii' codec can't encode character u'\xe1' in position 1:                                                                      ordinal not in range(128)

elParaguayo, 2016-01-12:
Thanks. I'll take a look. It may be possible to fix this just by changing the line to:
`evtime = u"{} - {}".format(evtime, event["location"])` 
but I'll try this when I get the chance.

elParaguayo, 2016-02-15:
This should work now. I'm working on some other changes and will add to list for next release.

closed 2016-02-20 by elParaguayo