elParaguayo/RPi-InfoScreen-Kivy #44: Squeezeplayer: Crash when Internet radio is playing

PaulWebster, 2016-02-15:
Seen crashes when Internet radio is playing - suspicion is that this is because there is no Album information present.

elParaguayo, 2016-02-15:
I've had some issues with the screen too. Will take a look.

Can you post the error message here too. Thanks.

PaulWebster, 2016-02-15:
Traceback (most recent call last):
   File "main.py", line 52, in <module>
     InfoScreenApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 828, in run
     runTouchApp()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 487, in runTouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 89, in mainloop
     self._mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 84, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 327, in idle
     Clock.tick()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 512, in tick
     self._process_events()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 644, in _process_events
     event.tick(self._last_tick, remove)
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 403, in tick
     ret = callback(self._dt)
   File "./screens/squeezeplayer/screen.py", line 804, in update
     self.checkForPlayers()
   File "./screens/squeezeplayer/screen.py", line 717, in checkForPlayers
     self.createPlayerScreen()
   File "./screens/squeezeplayer/screen.py", line 761, in createPlayerScreen
     sq_root=self)
   File "./screens/squeezeplayer/screen.py", line 144, in __init__
     super(SqueezeNowPlaying, self).__init__(*args, **kwargs)
   File "/usr/local/lib/python2.7/dist-packages/kivy/uix/accordion.py", line 361, in __init__
     super(Accordion, self).__init__(**kwargs)
   File "/usr/local/lib/python2.7/dist-packages/kivy/uix/widget.py", line 320, in __init__
     Builder.apply(self, ignored_consts=self._kwargs_applied_init)
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 1993, in apply
     self._apply_rule(widget, rule, rule, ignored_consts=ignored_consts)
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 2154, in _apply_rule
     e), cause=tb)
 kivy.lang.BuilderException: Parser: File "<inline>", line 643:
 ...
     641:                        halign: "left"
     642:                        padding: 5, 2
 >>  643:                        text: root.cur_track["album"]
     644:
     645:                    BoxLayout:
 ...
 BuilderException: Parser: File "<inline>", line 643:
 ...
     641:                        halign: "left"
     642:                        padding: 5, 2
 >>  643:                        text: root.cur_track["album"]
     644:
     645:                    BoxLayout:
 ...
 KeyError: 'album'
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 1742, in create_handler
     return eval(value, idmap), bound_list
   File "<string>", line 643, in <module>

   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 2139, in _apply_rule
     rctx['ids'])
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 1747, in create_handler
     cause=tb)

elParaguayo, 2016-02-15:
Thank you. I'll fix this when I get the chance.

elParaguayo, 2016-02-18:
Have a look at this branch and see if it works for you: https://github.com/elParaguayo/RPi-InfoScreen-Kivy/compare/squeeze_bugfixes

PaulWebster, 2016-02-19:
Looks like there is still a problem

pi@raspberrypi ~/RPi-InfoScreen-Kivy $ git branch
  master
* squeeze_bugfixes
.
.
.
 Traceback (most recent call last):
   File "main.py", line 52, in <module>
     InfoScreenApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 828, in run
     runTouchApp()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 487, in runTouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 89, in mainloop
     self._mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 84, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 327, in idle
     Clock.tick()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 512, in tick
     self._process_events()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 644, in _process_events
     event.tick(self._last_tick, remove)
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 403, in tick
     ret = callback(self._dt)
   File "./screens/squeezeplayer/screen.py", line 795, in update
     self.checkForPlayers()
   File "./screens/squeezeplayer/screen.py", line 708, in checkForPlayers
     self.createPlayerScreen()
   File "./screens/squeezeplayer/screen.py", line 752, in createPlayerScreen
     sq_root=self)
   File "./screens/squeezeplayer/screen.py", line 144, in __init__
     super(SqueezeNowPlaying, self).__init__(*args, **kwargs)
   File "/usr/local/lib/python2.7/dist-packages/kivy/uix/accordion.py", line 361, in __init__
     super(Accordion, self).__init__(**kwargs)
   File "/usr/local/lib/python2.7/dist-packages/kivy/uix/widget.py", line 320, in __init__
     Builder.apply(self, ignored_consts=self._kwargs_applied_init)
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 1993, in apply
     self._apply_rule(widget, rule, rule, ignored_consts=ignored_consts)
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 2154, in _apply_rule
     e), cause=tb)
 kivy.lang.BuilderException: Parser: File "<inline>", line 221:
 ...
     219:                        halign: "left"
     220:                        padding: 5, 2
 >>  221:                        text: root.cur_track["album"]
     222:
     223:                    BoxLayout:
 ...
 BuilderException: Parser: File "<inline>", line 221:
 ...
     219:                        halign: "left"
     220:                        padding: 5, 2
 >>  221:                        text: root.cur_track["album"]
     222:
     223:                    BoxLayout:
 ...
 KeyError: 'album'
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 1742, in create_handler
     return eval(value, idmap), bound_list
   File "<string>", line 221, in <module>

   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 2139, in _apply_rule
     rctx['ids'])
   File "/usr/local/lib/python2.7/dist-packages/kivy/lang.py", line 1747, in create_handler
     cause=tb)

elParaguayo, 2016-02-19:
That's weird. The error is showing `text: root.cur_track["album"]` but the updated code (in the squeeze_bugfixes branch) has `text: root.album_name`.

Have you definitely pulled down the right branch?

PaulWebster, 2016-02-19:
You are right - and very strange.
I included the output from "git branch" at the top of my previous post to show that it was the selected one. I just checked the .kv file and saw that the album definition did not have the change as above - so I deleted it all and started from scratch with a clone of just that branch. I could then see the commented out line and the replacement.
It now loads up OK.

It does have some issues though - for example
[ERROR  ] [Image       ] Error reading file /sq_volume.png
[ERROR  ] [Image       ] Error reading file /sq_next.png
[ERROR  ] [Image       ] Error reading file /sq_stop.png
[ERROR  ] [Image       ] Error reading file /sq_play.png
[ERROR  ] [Image       ] Error reading file /sq_previous.png

which means that the icons for controlling the player do not display. Should they have a /icons in front of them?

Also had this when a track was playing
[ERROR  ] [Loader      ] Failed to load image <http://www.mysqueezebox.com/public/imageproxy?h=50&u=plugins%2FTuneIn%2Fhtml%2Fimages%2Ficon.png&w=50>

and the volume slider was not representing what the volume was really set to.
When I changed the volume using a mobile phone app the volume jumped up and then went back down to 0 on the Kivy application.

elParaguayo, 2016-02-19:
I do apologise. This is proving quite painful, isn't it?!

I've seen the image error before, will fix that.

The volume one is interesting, let me check that. I've recently upgraded my server and had seen my volume at nil too. I wonder if they changed the protocol slightly.

Pretty sure I know what the image issue is too. I think I need to change the default method for non-local media.

I'll open these as separate issues.

elParaguayo, 2016-02-20:
Closing this thread as the original issue has been fixed.

closed 2016-02-20 by elParaguayo