elParaguayo/RPi-InfoScreen-Kivy #63: Is a Conning page with OpenGL okay?

karrika, 2016-03-05:
Hi elParaguayo,

I am quite interested in sailing and I would like to build a set of old fashioned, modern conning displays. The new Raspberry Pi OpenGL works very well. The question is just if you feel it is ok to use your repository for this.

Have a look at [the compass](http://79.125.115.174/pics/SlimCompass.png) to see what I mean. I also have a video clip of the compass in action [here](http://79.125.115.174/pics/00_slimCompass.mpg). All my sensors like wind, compass, drift etc would by accessible with up/down. They would monitor the IEC 61162 450 multicast streams making it plug-and-play to any modern vessel.

This would be usable on a real vessel.

For Raspberry Pi Sense HAT's I could also include a python relay module that listens to NMEA serial streams and relays the stuff as multicasts to the displays.

elParaguayo, 2016-03-05:
I'm a sailor too. Much smaller boats though!

Quite frankly, if the screen works, I'm happy to include it. The compass looks amazing.

karrika, 2016-03-05:
Great! It will take a while to do it as my old code written years ago was C++. The compass mimics Gyro displays found on SOLAS vessels. But I have a few nice new kind of displays also. What I have in mind is to create 3 screens. The first is compass + next waypoint + cross track error. The second would be wind + trend. The third is speed, drift and propulsion.

I hope to make the screens simple enough to be usable. I really hate zillions of digital displays showing just numbers. But I would also like to include special elements like "acceleration/deceleration" to show how many knots/min we are slowing down. This kind of info is important when you need to pick up a pilot on board. On a small boat you could use this for trimming sails also.

My only concern is that all these screens will be heavily animated. Without OpenGL they would be useless. But with OpenGL I believe they will be just beautiful. So Pi 2 and Pi 3 will be the only ones able to run these screens.

elParaguayo, 2016-03-05:
It's probably worth doing a "proof of concept" to check that you can get openGL working in a screen.

I do agree, it looks beautiful. I'd love to know how to code something like that.

closed 2016-12-10 by elParaguayo