elParaguayo/RPi-InfoScreen-Kivy #67: Digital clock enhancement

hybriddave, 2016-04-22:
Not sure if development on this project has ceased but thought I'd request an add on to the digital clock. I have scoured the internet for something like this and have installed this and it is working great. I am new to kivy and python but an alarm feature would be a good add on. Perhaps two alarm seperate alarm times.

elParaguayo, 2016-04-22:
Development has slowed on this as the bits that I use are working as I need. I am also working on a separate project at the moment. 

That said, if there are bugs I'll try to fix them. 

Feature requests are slightly different. My preference would be for someone else to code it and then I can merge it into the repo. The idea was to demonstrate what the script can do and allow new users to learn some coding by creating/changing screens.

hybriddave, 2016-04-22:
Thank you so much for fast response and for the direction. I look forward
to trying to add to this project and learning along the way.

Best regards,
David

closed 2016-12-10 by elParaguayo