elParaguayo/RPi-InfoScreen-Kivy #68: Photoalbum crashes

elParaguayo, 2016-04-22:
Reported on forum: https://www.raspberrypi.org/forums/viewtopic.php?f=41&t=121392&p=959154#p956896

kivitalo, 2016-08-03:
Hi, 
I was able to make some progress in this. If you increase the gpu memory allocated for the Raspberry PI the photoalbum is able to show the pictures. I still see some glError messages in the log but the pictures are shown correctly.
I will do some more testing with larger amount of pictures so I can see if this is really working and if there are any stability issues.

You can check the available memory for arm and gpu with the `vcgencmd get_mem arm && vcgencmd get_mem gpu` command in the commandline.
I had in my Raspberry Pi2 by default 64M allocated for the GPU so I increased that to 512M.
This can be done in the /boot/config.txt file by adding/modifing the `gpu_mem` keyword.
In my case I added `gpu_mem=512` to the end of the /boot/config.txt file.
Depending on the Raspberry Pi model the value might different.
I will get back once I have more results.

kivitalo, 2016-08-18:
Those glError messages are coming from the ScreenManager Transition effects. Some of the images are not show at all. If I set NoTransition images are show correctly and no glErrors. I am now testing with lot of images to see if there are some issues still. I don't know what would be the best way to solve this. Maybe first load to memory and resize to smaller ( Raspberry screen sized ) image and then show with Transitions. Also I don't know if there are any settings in the Kivy side related to OpenGL and transitions.

elParaguayo, 2016-08-18:
Thanks. It would be a shame to lose the transitions as otherwise a sudden change of image can be quite stark.

I know nothing about OpenGL I'm afraid.