elParaguayo/RPi-InfoScreen-Kivy #70: Word Clock fails to start if Xmas enabled

jimbolaya, 2016-05-23:
If I enable Xmas, when it scrolls over to Word Clock the "Word Clock" text in the middle of the screen just vibrates back and forth and the display is stuck on that screen. No touching the right or left side does anything by make the text vibrate.

When Xmas is disabled, the Word Clock function returns. 

I don't see any extra text on the console when this happens.

elParaguayo, 2016-05-24:
That's weird.  I've not seen that behavior.

I can try to take a look but it really won't be any time soon as I don't have much time for this project at the moment.

elParaguayo, 2016-12-10:
Works ok for me.

karrika, 2016-12-18:
Works ok for me on a Raspberry Pi 2 + 7" Official display.
I have xmas, wordclock(finnish), clock and finlandarrivals running