elParaguayo/RPi-InfoScreen-Kivy #73: only see Qlocktwo clock

wagi007, 2016-06-14:
hello as the title say i only see the clock in words, but whe i touch nothing will happen, i get only 1 warning that Bottle module could not start, touch works fine for all other kivy scripts. any clue what i am doing wrong?

elParaguayo, 2016-06-14:
When you say "Kivy works fine for all other kivy scripts" what do you mean? 

By default, the only screen that is enabled is the wordclock. As there are no other screens enabled, there are no screens that can be navigated to by touch and so touching the screen will do nothing.

Have you enabled other screens? If not, you've got two options: 1) manually enable the ones you want (by editing the screen.json file) or 2) install bottle and use the web interface.

wagi007, 2016-06-14:
hello thank u for ur fast answere :) i see now in the readme >.< noob fault by me but my english is horrible and i don't saw it before.

nice project

closed 2016-12-10 by elParaguayo