elParaguayo/RPi-InfoScreen-Kivy #77: ISS HD Stream

thegnnu, 2016-10-08:
Hi I see you have 1 screen showing the ISS track do you have any idea if you will be doing a HD stream that is now on line and only take one line of bash to give a simple streaming screen.
TerryR

elParaguayo, 2016-10-08:
I'm not actively working on this code at the moment. However, if you can get it working, I'd happily merge it into the main code.

thegnnu, 2016-10-10:
I was asking as I do  not have the PI Screen but would get one if I could use it for monitoring the ISS Stream. Do not have any use for the transport screens as long ago retired.

elParaguayo, 2016-12-10:
Try this branch: https://github.com/elParaguayo/RPi-InfoScreen-Kivy/tree/iss_video

You'll need to enable the screen.

It works on my laptop, but I haven't tested on a pi.

closed 2017-05-10 by elParaguayo