elParaguayo/RPi-InfoScreen-Kivy #84: ImportError: No module named httplib2

elisseee, 2016-12-14:
Hello. I am new to Python but have an experience with other Object Oriented Languages. In any case, I am trying to run your project after installing Kivy. 
1. I have installed your project here: /home/pi/kivy/RPi-InfoScreen-Kivy
2. Was able to receive a secret token from Google Calendar APIs.
3. Getting error at this moment:
pi@raspberrypi:~ $ python /home/pi/kivy/RPi-InfoScreen-Kivy/screens/agenda/authorise.py
Traceback (most recent call last):
  File "/home/pi/kivy/RPi-InfoScreen-Kivy/screens/agenda/authorise.py", line 2, in <module>
    import httplib2
ImportError: No module named httplib2

Could it be that Kivy version is different now than in main.py file? I have changed a version line there to "1.9.0" maybe should have to "1.9.1".
Thank you in advance for your help.
Additional info -Raspbian Jesse with all updates

elParaguayo, 2016-12-15:
If you've got pip installed you could do `pip install httplib2 --upgrade`

elisseee, 2016-12-15:
Thank you, that worked. 
After installation it asked me about _apiclient_ and _discovery_ packages that I have installed also but still have an issue with this:
sudo python /home/pi/kivy/RPi-InfoScreen-Kivy/screens/agenda/authorise.py
Traceback (most recent call last):
  File "/home/pi/kivy/RPi-InfoScreen-Kivy/screens/agenda/authorise.py", line 6, in <module>
    from apiclient import discovery
ImportError: cannot import name discovery

I understand that you do not have to debug all these steps for me but /agenda folder is important because I have to have a Calendar running but stumbling every step after Kivy installation. Thank you

elParaguayo, 2016-12-15:
`pip install --upgrade google-api-python-client`

elisseee, 2016-12-15:
from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead

still...

elParaguayo, 2016-12-15:
That's one I haven't seen before! Try `pip install --upgrade pip`

elisseee, 2016-12-15:
At least something interesting for you))
That is what I got after running pip install --upgrade pip

sudo pip install --upgrade pip
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead

elParaguayo, 2016-12-15:
Do a search for that error on google and look for an answer on stackoveflow. It's not an error with my code so it's hard for me to help.

elisseee, 2016-12-15:
thank you. I will continue my battle.)

elisseee, 2016-12-15:
One more question and I hope last one.
The whole project structure should be like this /home/pi/kivy/RPi-InfoScreen-Kivy
or like that: /home/pi/kivy/**kivy**/RPi-InfoScreen-Kivy

elParaguayo, 2016-12-15:
It shouldn't matter.

elisseee, 2016-12-15:
Got this far...

File "/home/pi/kivy/RPi-InfoScreen-Kivy/screens/agenda/authorise.py", line 55, in <module>
    get_credentials()
  File "/home/pi/kivy/RPi-InfoScreen-Kivy/screens/agenda/authorise.py", line 42, in get_credentials
    store = oauth2client.file.Storage(credential_path)
AttributeError: 'module' object has no attribute 'file'

elParaguayo, 2016-12-15:
Did you install the Google module i mentioned earlier?

elisseee, 2016-12-15:
Yes I did. I have been following every step from your instructions

elParaguayo, 2016-12-15:
Weird. Try adding this to the top of authorise.py underneath the import block
`from oauth2client import file`

elisseee, 2016-12-15:
It Run. Another error probably related to stackoverflow issue...

/usr/local/lib/python2.7/dist-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /root/.credentials/calendar-python-quickstart.json: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))

Your browser has been opened to visit:

    https://accounts.google.com/o/oauth2/auth?scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fcalendar.readonly&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&response_type=code&client_id=782979102144-ebpsu3kq0l7pdbbj1fu2m5qpucoec6e4.apps.googleusercontent.com&access_type=offline

If your browser is on a different machine then exit and re-run this
application with the command-line parameter
  --noauth_local_webserver

elisseee, 2016-12-15:
I meant I have to search stackoverflow and this is not your issue anymore. Thank you for all your patience and help!

elParaguayo, 2016-12-15:
Look at the readme in the agenda folder. You need to run authorise.py separately the first time.

elisseee, 2016-12-15:
Yes. Now it worked and had been authorised in a browser. Thank you!

closed 2017-05-10 by elParaguayo