elParaguayo/RPi-InfoScreen-Kivy #85: [WARNING] Bottle module not found. Cannot start webserver.

elisseee, 2016-12-16:
I am trying to run ./main.py and get the warning above. I have checked and see Bottle module installed and has a latest version.
Considering that maybe some dependencies are having issues I am going to put all my errors after trying to run ./main.py. 
`Traceback (most recent call last):
   File "./main.py", line 88, in <module>
     InfoScreenApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 828, in run
     runTouchApp()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 500, in runTouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 90, in mainloop
     self._mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 85, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 336, in idle
     Clock.tick()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 581, in tick
     self._process_events()
   File "kivy/_clock.pyx", line 368, in kivy._clock.CyClockBase._process_events (kivy/_clock.c:7740)
   File "kivy/_clock.pyx", line 398, in kivy._clock.CyClockBase._process_events (kivy/_clock.c:7617)
   File "kivy/_clock.pyx", line 396, in kivy._clock.CyClockBase._process_events (kivy/_clock.c:7538)
   File "kivy/_clock.pyx", line 168, in kivy._clock.ClockEvent.tick (kivy/_clock.c:3530)
 TypeError: create_window() takes exactly 1 argument (2 given)
`

elParaguayo, 2016-12-16:
The warning doesn't stop the program from loading (it's a warning, not an error) but it's strange that you get that message if bottle IDs installed.

Your actual error is likely because one screen is not working. Disable all of your screens except one. If the program works, enable another screen and run again until you find which screen is causing the problem.

elisseee, 2016-12-16:
They were disabled by default, except Worldclock. 
`"enabled": false` 
I have disabled Worldclock and enabled one in Agenda folder. Received error that you describing in instruction:
`[ERROR  ] Unmet dependencies for agenda screen. Skipping...
[ERROR  ] Unmet dependencies for agenda screen. Skipping...
`
I will continue enable one screen at a time and see, which one is causing all these problems.

elisseee, 2016-12-16:
so far from all modules error `Unmet dependencies for agenda screen. Skipping...` were in agenda, isstracker and mythtv (which is the only module that shows 3 Errors). I just don't know, which dependencies should be there?

elParaguayo, 2016-12-16:
No need to enable the screens that you don't want to use. I just meant you should disable all the screens you had previously enabled and then enable just those screens one by one.

I thought you had the agenda working. You shouldn't be getting an unmet dependency message if that's the case.

elisseee, 2016-12-16:
Any ideas how to move forward in this case? just want to make agenda working but I am stuck...

elParaguayo, 2016-12-16:
Disable all screens other than the agenda. If you still get the unmet dependency message then you need to install them. They are pytz and dateutil.

elisseee, 2016-12-16:
Yeah, I should have figure out from json file.. Anyway, I have installed these dependencies last error that i got is following: `[ERROR  ] Could not import agenda screen. Skipping...`

elParaguayo, 2016-12-16:
Enable the clock screen as well. You should get a message on your screen which should explain the agenda error.

elisseee, 2016-12-16:
`[ERROR  ] Could not import agenda screen. Skipping...`

elParaguayo, 2016-12-17:
I don't want the log message. What should happen is that, when a screen fails, it isn't displayed. Instead you should get a message displayed on the information screen itself. 

Can you disable the agenda and enable the wordclock and let me know if that works for you. If it does, then enable the agenda. The screen should show you a more detailed error message.

elisseee, 2016-12-19:
wordlclock was enabled. However, enabling agenda did not show me more detailed error message than I saw before.