elParaguayo/RPi-InfoScreen-Kivy #88: Energenie Screen

PVorster11, 2017-05-08:
This is awesome man! Struggling to get the Energenie screen to work. 

1. Hold down green button on Energenie plug untill it flashes
2. Hit "light" "on" button
3. Energenie plug light stops flashing 
4. On/off buttons do not trigger the energenie plug

Any help will be much appreciated,

Thank you!!

Phil﻿

elParaguayo, 2017-05-08:
My guess is that it's not the code. I had this issue a few times with plugs not registering the signals.

One hacky way that you could try is to send the on/off signal multiple times (rather than just the one). However, if you repeatedly tap the "on" button and the switch doesn't turn on then I very much doubt that that suggestion would help!

I've retired my energenie adapter as it interfered with some other lights in my house. This means I can't really test for you.

PVorster11, 2017-05-08:
Thank you for your prompt response!!!

I'm struggling!! Yes I agree, I doubt it is the code. 

I'm fairly sure I've done everything right. 

I've changed the host value to the IP address of my Pi (wlan0 inet addr)

I run PIGGPIO then run the program.

Loads up beautifully, the energenie adaptors react to intialisation but the buttons just wont trigger :-( 

Very sad as I need this working by the end of the week,

Thank you for your help!!

elParaguayo, 2017-05-08:
Try getting the switches to work using a basic script (ie forget about the Kivy screen for the moment).

Give me five mins and I'll send you some code to try.

PVorster11, 2017-05-08:
Thank you, thank you, thank you!!!

The Energenie plugs work perfectly when I run them using 👍 

#import the required modules
import RPi.GPIO as GPIO
import time

# set the pins numbering mode
GPIO.setmode(GPIO.BOARD)

# Select the GPIO pins used for the encoder K0-K3 data inputs
GPIO.setup(11, GPIO.OUT)
GPIO.setup(15, GPIO.OUT)
GPIO.setup(16, GPIO.OUT)
GPIO.setup(13, GPIO.OUT)

# Select the signal to select ASK/FSK
GPIO.setup(18, GPIO.OUT)

# Select the signal used to enable/disable the modulator
GPIO.setup(22, GPIO.OUT)

# Disable the modulator by setting CE pin lo
GPIO.output (22, False)

# Set the modulator to ASK for On Off Keying 
# by setting MODSEL pin lo
GPIO.output (18, False)

# Initialise K0-K3 inputs of the encoder to 0000
GPIO.output (11, False)
GPIO.output (15, False)
GPIO.output (16, False)
GPIO.output (13, False)

# The On/Off code pairs correspond to the hand controller codes.
# True = '1', False ='0'

print "OUT OF THE BOX: Plug the Pi Transmitter board into the Raspberry Pi"
print "GPIO pin-header ensuring correct polarity and pin alignment."
print ""
print "The sockets will need to be inserted into separate mains wall sockets."
print "with a physical separation of at least 2 metres to ensure they don't"
print "interfere with each other. Do not put into a single extension lead."
print ""
print "For proper set up the sockets should be in their factory state with"
print "the red led flashing at 1 second intervals. If this is not the case for"
print "either socket, press and hold the green button on the front of the unit"
print "for 5 seconds or more until the red light flashes slowly."
print ""
print "A socket in learning mode will be listening for a control code to be"
print "sent from a transmitter. A socket can pair with up to 2 transmitters"
print "and will accept the following code pairs"
print ""
print "0011 and 1011 all sockets ON and OFF"
print "1111 and 0111 socket 1 ON and OFF"
print "1110 and 0110 socket 2 ON and OFF"
print "1101 and 0101 socket 3 ON and OFF"
print "1100 and 0100 socket 4 ON and OFF"
print ""
print "A socket in learning mode should accept the first code it receives"
print "If you wish the sockets to react to different codes, plug in and"
print "program first one socket then the other using this program."
print ""
print "When the code is accepted you will see the red lamp on the socket"
print "flash quickly then extinguish"
print ""
print "The program will now loop around sending codes as follows when you"
print "hit a key:"
print "socket 1 on"
print "socket 1 off"
print "socket 2 on"
print "socket 2 off"
print "all sockets on"
print "all sockets off"
print "Hit CTL C for a clean exit"
try:
	# We will just loop round switching the units on and off
	while True:
		raw_input('hit return key to send socket 1 ON code')
		# Set K0-K3
		print "sending code 1111 socket 1 on"
		GPIO.output (11, True)
		GPIO.output (15, True)
		GPIO.output (16, True)
		GPIO.output (13, True)
		# let it settle, encoder requires this
		time.sleep(0.1)	
		# Enable the modulator
		GPIO.output (22, True)
		# keep enabled for a period
		time.sleep(0.25)
		# Disable the modulator
		GPIO.output (22, False)

		raw_input('hit return key to send socket 1 OFF code')
		# Set K0-K3
		print "sending code 0111 Socket 1 off"
		GPIO.output (11, True)
		GPIO.output (15, True)
		GPIO.output (16, True)
		GPIO.output (13, False)
		# let it settle, encoder requires this
		time.sleep(0.1)
		# Enable the modulator
		GPIO.output (22, True)
		# keep enabled for a period
		time.sleep(0.25)
		# Disable the modulator
		GPIO.output (22, False)

		raw_input('hit return key to send socket 2 ON code')
		# Set K0-K3
		print "sending code 1110 socket 2 on"
		GPIO.output (11, False)
		GPIO.output (15, True)
		GPIO.output (16, True)
		GPIO.output (13, True)
		# let it settle, encoder requires this
		time.sleep(0.1)	
		# Enable the modulator
		GPIO.output (22, True)
		# keep enabled for a period
		time.sleep(0.25)
		# Disable the modulator
		GPIO.output (22, False)

		raw_input('hit return key to send socket 2 OFF code')
		# Set K0-K3
		print "sending code 0110 socket 2 off"
		GPIO.output (11, False)
		GPIO.output (15, True)
		GPIO.output (16, True)
		GPIO.output (13, False)
		# let it settle, encoder requires this
		time.sleep(0.1)
		# Enable the modulator
		GPIO.output (22, True)
		# keep enabled for a period
		time.sleep(0.25)
		# Disable the modulator
		GPIO.output (22, False)

		raw_input('hit return key to send ALL ON code')
		# Set K0-K3
		print "sending code 1011 ALL on"
		GPIO.output (11, True)
		GPIO.output (15, True)
		GPIO.output (16, False)
		GPIO.output (13, True)
		# let it settle, encoder requires this
		time.sleep(0.1)
		# Enable the modulator
		GPIO.output (22, True)
		# keep enabled for a period
		time.sleep(0.25)
		# Disable the modulator
		GPIO.output (22, False)

		raw_input('hit return key to send ALL OFF code')
		# Set K0-K3
		print "sending code 0011 All off"
		GPIO.output (11, True)
		GPIO.output (15, True)
		GPIO.output (16, False)
		GPIO.output (13, False)
		# let it settle, encoder requires this
		time.sleep(0.1)	
		# Enable the modulator
		GPIO.output (22, True)
		# keep enabled for a period
		time.sleep(0.25)
		# Disable the modulator
		GPIO.output (22, False)

# Clean up the GPIOs for next time
except KeyboardInterrupt:
	GPIO.cleanup()

PVorster11, 2017-05-08:
Also, apologies if this is useless information but when I force the socket on by pressing the green button on the energenie plug, if i then press "on" or "off"  - kivy button, the socket responds by going low "0" so the socket is programmed successfully it just wont trigger a high "1"

elParaguayo, 2017-05-08:
Try this (I've made it trigger the switch five times). It's untested so there may be some stupid bugs.

```
# Module for Energenie switches using pigpio module
# All credit due to the original energenie.py scrip by Amy Mather
# See: https://github.com/MiniGirlGeek/energenie-demo/blob/master/energenie.py

import pigpio
from time import sleep

# The GPIO pins for the Energenie module
BIT1 = 17  # Board 11
BIT2 = 22  # Board 15
BIT3 = 16  # Board 16
BIT4 = 21  # Board 13

ON_OFF_KEY = 24  # Board 18
ENABLE = 25  # Board 22

# Codes for switching on and off the sockets
#        all     1       2       3       4
ON = ['1011', '1111', '1110', '1101', '1100']
OFF = ['0011', '0111', '0110', '0101', '0100']

class EnergenieControl(object):
    def __init__(self, **kwargs):
        self.host = kwargs.get("host", "")
        self.connect()
        self.setup()

    def connect(self):
        """Create an instance of the pigpio pi."""
        self.pi = pigpio.pi(self.host)

    def setup(self):
        """Clear the pin values before we use the transmitter."""
        if self.connected:
            self.pi.write(ON_OFF_KEY, False)
            self.pi.write(ENABLE, False)
            self.pi.write(BIT1, False)
            self.pi.write(BIT2, False)
            self.pi.write(BIT3, False)
            self.pi.write(BIT4, False)

    def __change_plug_state(self, socket, on_or_off):
        """Method to set up the pins and fire the transmitter."""
        state = on_or_off[socket][3] == '1'
        self.pi.write(BIT1, state)
        state = on_or_off[socket][2] == '1'
        self.pi.write(BIT2, state)
        state = on_or_off[socket][1] == '1'
        self.pi.write(BIT3, state)
        state = on_or_off[socket][0] == '1'
        self.pi.write(BIT4, state)
        sleep(0.1)
        self.pi.write(ENABLE, True)
        sleep(0.25)
        self.pi.write(ENABLE, False)

    def switch_on(self, socket):
        for _ in range(5):
             self.__change_plug_state(socket, ON)
             sleep(0.1)

    def switch_off(self, socket):
        for _ in range(5):
             self.__change_plug_state(socket, OFF)
             sleep(0.1)

    @property
    def connected(self):
        return self.pi.connected

if __name__ == "__main__":

    host = IP_ADDRESS_OF_YOUR_PI_HERE
    ec = EnergenieControl(host=host)

    while True:

        pin = raw_input("Enter number of pin to test or 'x' to exit: ")

	if pin in ["x", "X"]:
            break

	else:
            try:
                pin = int(pin)
                if pin < 1 or pin > 4:
                    print "Pin must be 1-4"
                    continue
	    except:
                print "Unexpected response."
                continue

        state = raw_input("Enter 1 to turn pin on or 0 for off: ")

        try:
            state = int(state)
            if state not in [0, 1]:
                print "Unexpected result"
                continue
        except:
            print "Unexpected result"
            continue

	if state == 1:
            ec.switch_on(pin)
        else:
            ec.switch_off(pin)
```

PVorster11, 2017-05-08:
The Energenie plugs work perfect with a simple script. It's only the kivy binding with which I think I am having problems

elParaguayo, 2017-05-08:
I don't think it's Kivy. That should just trigger the relevant function.

The code you said works uses the RPi.GPIO module whereas mine uses pigpio.

Did my example code work?

PVorster11, 2017-05-08:
Apologies, I should of realised that straight away. 

When I run the example code, I initialise "socket 1",  socket 1 successfully receives the signal. 

However switching socket 1 on or off with "1" or "0" does not do anything. 

Once again, when I force socket 1 to be on (green button), sending a "1" or "0" then triggers the socket to switch off.

elParaguayo, 2017-05-08:
Sending a 1 really shouldn't turn off the switch. That makes no sense to me at all.

PVorster11, 2017-05-08:
Exactly my thoughts.

PVorster11, 2017-05-08:
Is there a possibility that my PIGPIO isn't set up correctly?

elParaguayo, 2017-05-08:
Very unlikely. If the switch is reacting then the pins are working.

elParaguayo, 2017-05-08:
Is your energenie transmitter on a different pi? If it's on the same one then we could change the code to use RPi.GPIO.

PVorster11, 2017-05-08:
On the same Pi. My end goal is to have a GUI using Kivy to control 2 Energenie sockets. elParaguayo, honestly, I appreciate all of your help

elParaguayo, 2017-05-08:
Ok. If it's same pi then we can use RPi.GPIO.

Do you know enough python to adapt the code or do you need some help?

PVorster11, 2017-05-08:
I class myself as a complete novice! Some help would be wonderful! Thank you!!

elParaguayo, 2017-05-08:
No problem. I'll see if I can do this later today for you.

PVorster11, 2017-05-08:
Thank you!! Thank you!! Thank you!! I will have a go at it now, but not very confident.

elParaguayo, 2017-05-08:
Ok. Save this as "energenie_rpigpio.py" in screens/energenie:

```
# Module for Energenie switches using pigpio module
# All credit due to the original energenie.py scrip by Amy Mather
# See: https://github.com/MiniGirlGeek/energenie-demo/blob/master/energenie.py

import RPi.GPIO as GPIO
from time import sleep

# The GPIO pins for the Energenie module
BIT1 = 11
BIT2 = 15 
BIT3 = 16
BIT4 = 13 

ON_OFF_KEY = 18 
ENABLE = 22

# Codes for switching on and off the sockets
#        all     1       2       3       4
ON = ['1011', '1111', '1110', '1101', '1100']
OFF = ['0011', '0111', '0110', '0101', '0100']

class EnergenieControl(object):
    def __init__(self, **kwargs):
        self.setup()

    def setup(self):
        """Clear the pin values before we use the transmitter."""
        GPIO.setmode(GPIO.BOARD)
        GPIO.setwarnings(False)

        # Select the GPIO pins used for the encoder K0-K3 data inputs
        GPIO.setup(BIT1, GPIO.OUT)
        GPIO.setup(BIT2, GPIO.OUT)
        GPIO.setup(BIT3, GPIO.OUT)
        GPIO.setup(BIT4, GPIO.OUT)

        # Select the signal to select ASK/FSK
        GPIO.setup(ON_OFF_KEY8, GPIO.OUT)

        # Select the signal used to enable/disable the modulator
        GPIO.setup(ENABLE, GPIO.OUT)

        GPIO.output(ON_OFF_KEY, False)
        GPIO.output(ENABLE, False)
        GPIO.output(BIT1, False)
        GPIO.output(BIT2, False)
        GPIO.output(BIT3, False)
        GPIO.output(BIT4, False)

    def __change_plug_state(self, socket, on_or_off):
        """Method to set up the pins and fire the transmitter."""
        state = on_or_off[socket][3] == '1'
        GPIO.output(BIT1, state)
        state = on_or_off[socket][2] == '1'
        GPIO.output(BIT2, state)
        state = on_or_off[socket][1] == '1'
        GPIO.output(BIT3, state)
        state = on_or_off[socket][0] == '1'
        GPIO.output(BIT4, state)
        sleep(0.1)
        GPIO.output(ENABLE, True)
        sleep(0.25)
        GPIO.output(ENABLE, False)

    def switch_on(self, socket):
        self.__change_plug_state(socket, ON)

    def switch_off(self, socket):
        self.__change_plug_state(socket, OFF)
```
Then, in screen.py, change the line:
```from energenie_pigpio import EnergenieControl```
to
```from energenie_rpigpio import EnergenieControl```

PVorster11, 2017-05-08:
Hello,

You are a legend!! Thank you so so much.

After making the changes as well as changing the following:

# Select the signal to select ASK/FSK
        GPIO.setup(ON_OFF_KEY8, GPIO.OUT)

to 

       GPIO.setup(ON_OFF_KEY, GPIO.OUT)

I am getting an error (attatched as photo) apologies its not written code.

Thank you!
[image]

elParaguayo, 2017-05-08:
Oops. Sorry about that.

In the energenie_rpigpio.py script, change the __init__ function to this:
```
    def __init__(self, **kwargs):
        self.connected = True
        self.setup()
```

PVorster11, 2017-05-08:
THANK YOU THANK YOU THANK YOU!!!!! You do not understand how much you've helped me!!!

Your work is incredible, clean and easy to understand. THANK YOU!!

Now I just have to figure out how to get a temperature readings to update as a label on kivy.

elParaguayo, do you mind if acknowledge your work and name in my work?

elParaguayo, 2017-05-08:
You're very welcome! Happy to have been able to help.

PVorster11, 2017-05-08:
elParaguayo,  

I don't necessarily want to continue on this thread as the subject issue has been resolved. 

I was just wondering if you have any idea of how I would display the following as a label on kivy?

import time
import Adafruit_GPIO.SPI as SPI
import Adafruit_MAX31855.MAX31855 as MAX31855

########################################################
#### M A X 3 1 8 5 5 K   C O N F I G U R A T I O N ####
########################################################

# Define a function to convert celsius to fahrenheit.
def c_to_f(c):
        return c * 9.0 / 5.0 + 32.0

# Raspberry Pi software SPI configuration.
CLK_1 = 4
CS_1 = 3
DO_1 = 2
sensor1 = MAX31855.MAX31855(CLK_1, CS_1, DO_1)

CLK_2 = 25
CS_2 = 24
DO_2 = 23
sensor2 = MAX31855.MAX31855(CLK_2, CS_2, DO_2)

print('Press Ctrl-C to quit.')
while True:
    temp1 = sensor1.readTempC()
    temp2 = sensor2.readTempC()
    internal1 = sensor1.readInternalC()
    internal2 = sensor2.readInternalC()
    print('Thermocouple1 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(temp1, c_to_f(temp1)))
    # print('    Internal1 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(internal1, c_to_f(internal1)))
    print('Thermocouple2 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(temp2, c_to_f(temp2)))
    #print('    Internal2 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(internal2, c_to_f(internal2)))
    time.sleep(0.25)     # Loop printing measurements every second.

elParaguayo, 2017-05-09:
Start a new issue and I'll see if I have some time tonight to help.

closed 2017-05-09 by elParaguayo