elParaguayo/RPi-InfoScreen-Kivy #89: Temperature Readings to update on a kivy label

PVorster11, 2017-05-09:
Hi elParaguayo,

I was wondering if you could suggest a simple solution to get the following script to display in a kivy label.

Once again, I can't thank you enough for all of your help. 

Code:

import time
import Adafruit_GPIO.SPI as SPI
import Adafruit_MAX31855.MAX31855 as MAX31855

#Define a function to convert celsius to fahrenheit.
def c_to_f(c):
        return c * 9.0 / 5.0 + 32.0

#Raspberry Pi software SPI configuration.
CLK_1 = 13
CS_1 = 6
DO_1 = 5
sensor1 = MAX31855.MAX31855(CLK_1, CS_1, DO_1)

CLK_2 = 16
CS_2 = 26
DO_2 = 19
sensor2 = MAX31855.MAX31855(CLK_2, CS_2, DO_2)

print('Press Ctrl-C to quit.')
while True:
    temp1 = sensor1.readTempC()
    temp2 = sensor2.readTempC()
    internal1 = sensor1.readInternalC()
    internal2 = sensor2.readInternalC()
    print('Thermocouple1 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(temp1, c_to_f(temp1)))
    # print('    Internal1 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(internal1, c_to_f(internal1)))
    print('Thermocouple2 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(temp2, c_to_f(temp2)))
    #print('    Internal2 Temperature: {0:0.3F}*C / {1:0.3F}*F'.format(internal2, c_to_f(internal2)))
    time.sleep(1.0)     # Loop printing measurements every second.

elParaguayo, 2017-05-09:
Yes - should be pretty easy to do. I'll see if I can knock something together tonight.

elParaguayo, 2017-05-09:
Ok. Let's try this...

Create a new folder in "screens" called "temperature". Copy [this zip file](https://github.com/elParaguayo/RPi-InfoScreen-Kivy/files/987521/temperature.zip) into that folder and extract it.

Run the program and hope it works!

PVorster11, 2017-05-09:
Hello!

I get an error message saying the temperature screen has unmet 
dependencies: temperature: Adafruit_MAX31855

I take it this is from the .json file.

elParaguayo, 2017-05-09:
Strange. I would have expected that line to work if your temperature code works.

Change conf.json to this:
```
{
    "screen": "TemperatureScreen",
    "kv": "temperature.kv",
    "dependencies": [],
    "enabled": true
}
```

PVorster11, 2017-05-09:
I tried that and get:

Errors were encountered trying to create the following screens:
temperature: ValueError('A different mode has already been set".)

elParaguayo, 2017-05-09:
Sounds like Adafruit's code is conflicting with the Energenie code. Can you just try disabling the energenie screen for now.

elParaguayo, 2017-05-09:
If that works we can fix the energenie one!

PVorster11, 2017-05-09:
Can it be this include which is in my original code but not the new code?

import Adafruit_GPIO.SPI as SPI

PVorster11, 2017-05-09:
With the Energenie screen disabled, I still get the "unmet dependencies" error

elParaguayo, 2017-05-09:
No, won't be that line - your code doesn't use it directly.

Which dependencies are unmet?

PVorster11, 2017-05-09:
temperature: Adafruit_MAX31855

elParaguayo, 2017-05-09:
Sounds like you haven't updated conf.json

That line can only appear if "Adafruit_MAX31855" is listed in dependencies.

elParaguayo, 2017-05-09:
You'll need to restart the code after changing it.

PVorster11, 2017-05-09:
You're a genius!!! It works now.

PVorster11, 2017-05-09:
Now just figuring out why the Energenie screen doesn't work

elParaguayo, 2017-05-09:
OK - so we just need to fix Energenie (don't worry, it's easy)

elParaguayo, 2017-05-09:
In the energenie_rpigpio.py code, change the pin numbering lines to this:
```
BIT1 = 17
BIT2 = 22
BIT3 = 23
BIT4 = 27 

ON_OFF_KEY = 24
ENABLE = 25
```
and change this line:
```
GPIO.setmode(GPIO.BOARD)
```
to
```
GPIO.setmode(GPIO.BCM)
```

PVorster11, 2017-05-09:
WORKS 100%, Thank you!!!!!! What was the issue?

elParaguayo, 2017-05-09:
The RPi.GPIO module only lets you use one pin numbering layout (either board numbering or Broadcom numbering).

The temperature module forces the Broadcom numbering but my Energenie code was using board numbering. I've changed it to use the Broadcom numbering.

You can see the different numbering layouts [here](https://pinout.xyz/).

Incidentally, I think this is why the pigpio code didn't work, I think I got the numbers wrong!!

PVorster11, 2017-05-09:
Thank you once again, really appreciate all of your help. THANK YOU!!

closed 2017-05-10 by elParaguayo