elParaguayo/RPi-InfoScreen-Kivy #90: Displaying camera output in a window on a kivy "screen"

PVorster11, 2017-05-09:
Hi elParaguayo,

The last issue I have is displaying a live video feed in a window on a screen managed by screen manager (kivy).

Thank you!!!

code:

from picamera import PiCamera, Color
from time import sleep

camera = PiCamera()

camera.resolution = (1920,1080)
camera.framerate = 30
camera.start_preview()
camera.awb_mode = 'fluorescent'
camera.exposure_mode = 'auto'
camera.annotate_text = "1920x1080 30fr auto exposure awb-flourescent"
camera.annotate_text_size = 50
camera.annotate_background = Color('black')
camera.annotate_foreground = Color('white')
camera.start_recording('/home/pi/darren.h264')
sleep(10)
#camera.capture('/home/pi/Desktop/fools.jpg')
camera.stop_recording()
camera.stop_preview()

elParaguayo, 2017-05-09:
I know nothing about PiCamera and have very little experience with videos within Kivy so I may not be able to do much here.

You'd need to provide a source to the Kivy videoplayer and I don't know whether it can read the "darren.h264" file if it's being written to by the code above. The alternative would be to provide a stream address and have Kivy try to play that.

PVorster11, 2017-05-09:
I'll do some digging, thank you, that makes a lot of sense.

elParaguayo, 2017-05-09:
Have a look at this: https://www.raspberrypi.org/forums/viewtopic.php?t=122413&p=826925

PVorster11, 2017-05-09:
Hahaha, I was literally just on this page

closed 2017-05-10 by elParaguayo