elParaguayo/RPi-InfoScreen-Kivy #91: PhotoAlbum Error

PVorster11, 2017-05-14:
Hi elParaguayo,

I'm back, haaha!

I am getting the following error when, I think, I've follwed all of the instructions to set up the Photo Album screen.

Any help would be appreciated.

Thank you!

Phil
[image]

elParaguayo, 2017-05-14:
What's in the conf.json file?

PVorster11, 2017-05-14:
{
    "screen": "PhotoAlbumScreen",
    "kv": "photos.kv",
    "dependencies": [],
    "enabled": true,
    "params":
    {
        "duration": 10,
        "extensions": ["jpg", "png"],
        "folders":
        [
            "/home/PiTrude/images"
        ]
    }
}

elParaguayo, 2017-05-14:
And are there any images in that folder? It looks like the script can't find any photos.

PVorster11, 2017-05-14:
Yes, 3 photos.

PVorster11, 2017-05-14:
My fault!!, Directory should be /home/pi/PiTrude.....

PVorster11, 2017-05-14:
Works now but just getting the white spinning wheel of death

elParaguayo, 2017-05-14:
Anything unusual about the images? Special characters in filename?

PVorster11, 2017-05-14:
10x10_transparent.png, plastics1.jpg, plasticfacts.jpg

elParaguayo, 2017-05-14:
Strange. Can't think of any reason for it not to work.