elParaguayo/RPi-InfoScreen-Kivy #94: not working in Python3...

ckwjdals90, 2017-06-28:
Hello,
Thanks for sharing the awesome project!
I am fairly new to Python world and only a week old of kivy stuffs, so if I do not understand things well, please bare with me :P
I am trying to learn from your code and create my own dashboard.
I first ran it in Python 2.7, it was working very well, then I tried to run it with Python 3.6 (since I am planning to create one with 3.6) it gave me this error:

> [INFO   ] [Logger      ] Record log in /Users/ckwjdals90/.kivy/logs/kivy_17-06-28_46.txt
> [INFO   ] [Kivy        ] v1.10.0
> [INFO   ] [Python      ] v3.6.1 (default, Apr  4 2017, 09:40:21) 
> [GCC 4.2.1 Compatible Apple LLVM 8.1.0 (clang-802.0.38)]
> [INFO   ] [Factory     ] 194 symbols loaded
> [INFO   ] [Image       ] Providers: img_tex, img_imageio, img_dds, img_sdl2, img_pil, img_gif (img_ffpyplayer ignored)
> [INFO   ] [OSC         ] using <multiprocessing> for socket
> [INFO   ] [Window      ] Provider: sdl2
> [INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
> [INFO   ] [GL          ] Backend used <gl>
> [INFO   ] [GL          ] OpenGL version <b'2.1 ATI-1.51.8'>
> [INFO   ] [GL          ] OpenGL vendor <b'ATI Technologies Inc.'>
> [INFO   ] [GL          ] OpenGL renderer <b'AMD Radeon Pro 460 OpenGL Engine'>
> [INFO   ] [GL          ] OpenGL parsed version: 2, 1
> [INFO   ] [GL          ] Shading version <b'1.20'>
> [INFO   ] [GL          ] Texture max size <16384>
> [INFO   ] [GL          ] Texture max units <16>
> [INFO   ] [Window      ] auto add sdl2 input provider
> [INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
> [INFO   ] [Text        ] Provider: sdl2
> [WARNING] Bottle module not found. Cannot start webserver.
>  Traceback (most recent call last):
>    File "main.py", line 90, in <module>
>      InfoScreenApp().run()
>    File "/usr/local/lib/python3.6/site-packages/kivy/app.py", line 802, in run
>      root = self.build()
>    File "main.py", line 37, in build
>      self.base = InfoScreen(plugins=plugins)
>    File "/Users/ckwjdals90/project/python/kvtuts/rpi_infoscreen_kv/core/infoscreen.py", line 19, in __init__
>      super(InfoScreen, self).__init__(**kwargs)
>    File "/usr/local/lib/python3.6/site-packages/kivy/uix/floatlayout.py", line 65, in __init__
>      super(FloatLayout, self).__init__(**kwargs)
>    File "/usr/local/lib/python3.6/site-packages/kivy/uix/layout.py", line 76, in __init__
>      super(Layout, self).__init__(**kwargs)
>    File "/usr/local/lib/python3.6/site-packages/kivy/uix/widget.py", line 337, in __init__
>      super(Widget, self).__init__(**kwargs)
>    File "kivy/_event.pyx", line 254, in kivy._event.EventDispatcher.__init__ (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:5332)
>  TypeError: object.__init__() takes no parameters

Could you please help me out?

elParaguayo, 2017-06-29:
Yes. Don't use python 3!

It wasn't written with python 3 in mind so I have no idea if it will work.

If you really want to try it then I would disable all screens except the clock and see if that works.

If it does then try enabling other screens one by one and see if they work.

ckwjdals90, 2017-06-29:
hmm.. I see

I did try to run the app with everything disabled except for the clock, and still gave me the same error.

Any other suggestions I could try out?

Thank you :D

elParaguayo, 2017-06-29:
Is there any reason you have to use python 3?

ckwjdals90, 2017-06-29:
Our lead wants us to use Python 3, and I love your work and structure of codes, so I want to learn to use it haha

elParaguayo, 2017-06-29:
Well, your lead is right. All new code should be in python 3 unless there's a good reason not to use it.

Unfortunately, I learnt python 2 (as that's what I needed to develop some (legitimate) add-ons for xbmc/kodi).

To be honest, I won't have the time to see if I could get it to run in python 3.

ckwjdals90, 2017-06-29:
That is totally understandable.

I will try and see what I can do to make it work in Python 3.

If you have insight where I might want to check out, please let me know.

And again, thank you so much for sharing the awesome project, and sharing your thoughts.

Cheers

ckwjdals90, 2017-07-06:
Hello again!

Sorry to bother you again D:

I figured that it was how super() works in python3, so I managed to get it to work!

However, I have another issue when I try to display the next screen (in this case, "clock" again since the only active screen is the "clock")

`[INFO   ] [Logger      ] Record log in /Users/ckwjdals90/.kivy/logs/kivy_17-07-06_90.txt
[INFO   ] [Kivy        ] v1.10.0
[INFO   ] [Python      ] v3.6.1 (default, Apr  4 2017, 09:40:21) 
[GCC 4.2.1 Compatible Apple LLVM 8.1.0 (clang-802.0.38)]
[INFO   ] [Factory     ] 194 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_imageio, img_dds, img_sdl2, img_pil, img_gif (img_ffpyplayer ignored)
[INFO   ] [OSC         ] using <multiprocessing> for socket
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <b'2.1 ATI-1.51.8'>
[INFO   ] [GL          ] OpenGL vendor <b'ATI Technologies Inc.'>
[INFO   ] [GL          ] OpenGL renderer <b'AMD Radeon Pro 460 OpenGL Engine'>
[INFO   ] [GL          ] OpenGL parsed version: 2, 1
[INFO   ] [GL          ] Shading version <b'1.20'>
[INFO   ] [GL          ] Texture max size <16384>
[INFO   ] [GL          ] Texture max units <16>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[WARNING] Bottle module not found. Cannot start webserver.
[INFO   ] [Screen      ] clock loaded.
[INFO   ] [Base        ] Start application main loop
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "main.py", line 74, in <module>
     InfoScreenApp().run()
   File "/usr/local/lib/python3.6/site-packages/kivy/app.py", line 828, in run
     runTouchApp()
   File "/usr/local/lib/python3.6/site-packages/kivy/base.py", line 504, in runTouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python3.6/site-packages/kivy/core/window/window_sdl2.py", line 663, in mainloop
     self._mainloop()
   File "/usr/local/lib/python3.6/site-packages/kivy/core/window/window_sdl2.py", line 405, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python3.6/site-packages/kivy/base.py", line 342, in idle
     self.dispatch_input()
   File "/usr/local/lib/python3.6/site-packages/kivy/base.py", line 327, in dispatch_input
     post_dispatch_input(*pop(0))
   File "/usr/local/lib/python3.6/site-packages/kivy/base.py", line 233, in post_dispatch_input
     listener.dispatch('on_motion', etype, me)
   File "kivy/_event.pyx", line 718, in kivy._event.EventDispatcher.dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:8191)
   File "/usr/local/lib/python3.6/site-packages/kivy/core/window/__init__.py", line 1188, in on_motion
     self.dispatch('on_touch_down', me)
   File "kivy/_event.pyx", line 718, in kivy._event.EventDispatcher.dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:8191)
   File "/usr/local/lib/python3.6/site-packages/kivy/core/window/__init__.py", line 1204, in on_touch_down
     if w.dispatch('on_touch_down', touch):
   File "kivy/_event.pyx", line 718, in kivy._event.EventDispatcher.dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:8191)
   File "/usr/local/lib/python3.6/site-packages/kivy/uix/widget.py", line 457, in on_touch_down
     if child.dispatch('on_touch_down', touch):
   File "kivy/_event.pyx", line 718, in kivy._event.EventDispatcher.dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:8191)
   File "/usr/local/lib/python3.6/site-packages/kivy/uix/behaviors/button.py", line 151, in on_touch_down
     self.dispatch('on_press')
   File "kivy/_event.pyx", line 714, in kivy._event.EventDispatcher.dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:8146)
   File "kivy/_event.pyx", line 1225, in kivy._event.EventObservers.dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:14035)
   File "kivy/_event.pyx", line 1109, in kivy._event.EventObservers._dispatch (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/_event.c:12816)
   File "/usr/local/lib/python3.6/site-packages/kivy/lang/builder.py", line 64, in custom_callback
     exec(__kvlang__.co_value, idmap)
   File "/Users/ckwjdals90/project/python/kvtuts/testbench_kv/dashboard_kv/base.kv", line 16, in <module>
     on_press: root.next_screen()
   File "/Users/ckwjdals90/project/python/kvtuts/testbench_kv/dashboard_kv/core/infoscreen.py", line 192, in next_screen
     self.scrmgr.current = self.availablescreens[self.index]
   File "kivy/weakproxy.pyx", line 33, in kivy.weakproxy.WeakProxy.__setattr__ (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/weakproxy.c:1471)
   File "kivy/properties.pyx", line 478, in kivy.properties.Property.__set__ (/private/var/folders/5s/l2s9hpz92bn6nr2c5239rvm80000gn/T/pip-build-rkjslunl/kivy/kivy/properties.c:5572)
[... cut ...]
   File "/usr/local/lib/python3.6/site-packages/kivy/uix/screenmanager.py", line 1037, in on_current
     screen = self.get_screen(value)
   File "/usr/local/lib/python3.6/site-packages/kivy/uix/screenmanager.py", line 1063, in get_screen
     raise ScreenManagerException('No Screen with name "%s".' % name)
 kivy.uix.screenmanager.ScreenManagerException: No Screen with name "clock".

shell returned 1`

it is super weird because I see the clock screen working and displayed, but as soon as I click on the hidden button, it throws this error message.

Any idea why it is breaking??

karrika, 2019-05-17:
I found out that if you add an extra class to all screens like:

```
class BlackHole(object):
    def __init__(self, **kw):
        super(BlackHole, self).__init__()

class FinlandArrivals(BoxLayout, BlackHole):

```
Then the stuff works with python3. As a lot of my pull requests are still hanging I have not yet made a pull request that would make this work for python3.

elParaguayo, 2019-05-17:
Sorry everyone! I haven't looked at this project in a lot for time and I haven't been writing stuff for the pi either. 

Karrika, great that you've got it working on python 3.

I don't have time to test pull requests but I am happy to merge. Let me know if you want to update them for python3 before I merge. 

I should also update to say that this project is unmaintained!

Thanks for your interest in this project.

karrika, 2019-05-17:
Thanks for returning on this. I could spend some time to test the pull requests and create a python3 pull as well. I am also running this on an Ubuntu 19.04 workstation. And plan to try it on Windows as well.

elParaguayo, 2019-05-17:
Ok. Let me merge them as they are and, if you do update for python 3, just submit a new pull request.

karrika, 2019-05-17:
Sounds great!